uzh-rpg/agile_flight #9: Error -roslaunch unable to launch [kingfisher/dodgeros_pilot-2]

Tsuji-Kosei, 2022-03-21:
I faced with the following error when I typed "roslaunch envsim visionenv_sim.launch render:=True"

```
... logging to /home/user/.ros/log/9b8562be-a8f9-11ec-8908-f9ada68fd0d8/roslaunch-user-ubuntu-3814.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://user-ubuntu:35019/

SUMMARY
========

PARAMETERS
 * /kingfisher/dodgeros_pilot/agi_param_dir: /home/user/icra2...
 * /kingfisher/dodgeros_pilot/camera_config: /home/user/icra2...
 * /kingfisher/dodgeros_pilot/low_level_controller: Simple
 * /kingfisher/dodgeros_pilot/pilot_config: simple_sim_pilot....
 * /kingfisher/dodgeros_pilot/real_time_factor: 1.0
 * /kingfisher/dodgeros_pilot/render: True
 * /kingfisher/dodgeros_pilot/ros_param_dir: /home/user/icra2...
 * /kingfisher/dodgeros_pilot/use_bem_propeller_model: False
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /use_sim_time: True

NODES
  /
    dodgeros_gui (rqt_gui/rqt_gui)
    flight_render (flightrender/RPG_Flightmare.x86_64)
  /kingfisher/
    dodgeros_pilot (envsim/visionsim_node)
    viz_face (rviz/rviz)

auto-starting new master
process[master]: started with pid [3822]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9b8562be-a8f9-11ec-8908-f9ada68fd0d8
process[rosout-1]: started with pid [3832]
started core service [/rosout]
process[kingfisher/dodgeros_pilot-2]: started with pid [3839]
process[kingfisher/viz_face-3]: started with pid [3840]
process[dodgeros_gui-4]: started with pid [3841]
process[flight_render-5]: started with pid [3842]
[ WARN] [1647855055.853602462]: Pilot Config:        simple_sim_pilot.yaml
[ WARN] [1647855055.854849416]: Agi Param Directory: /home/user/icra22_competition_ws/src/agile_flight/dodgedrone_simulation/dodgelib/params
[ WARN] [1647855055.854895142]: ROS Param Directory: /home/user/icra22_competition_ws/src/agile_flight/envsim/parameters
[ INFO] [1647855055.854923451]: Loading Pilot Params from simple_sim_pilot.yaml in /home/user/icra22_competition_ws/src/agile_flight/envsim/parameters
Loading Pilot parameters from "/home/user/icra22_competition_ws/src/agile_flight/envsim/parameters/simple_sim_pilot.yaml"
[PilotParams]  Did not create inner controller ''!
[Pilot]        Did not create bridge 'ROS'.
Using debug bridge, register externally!
[ INFO] [1647855055.860944218]: Loaded pipeline:
Estimator:
Type: Feedthrough
File: "/home/user/icra22_competition_ws/src/agile_flight/dodgedrone_simulation/dodgelib/params/feedthrough.yaml"
Sampler:
Type: Time
File: ""
Outer Controller:
Type: GEO
File: "/home/user/icra22_competition_ws/src/agile_flight/dodgedrone_simulation/dodgelib/params/geo.yaml"
Inner Controller:
Type: 
File: ""
Bridge:
Type: ROS
File: ""

[Pilot]        Register external bridge: [ROS Bridge]   
 which was not active and used.
[kingfisher/dodgeros_pilot-2] process has died [pid 3839, exit code -11, cmd /home/user/icra22_competition_ws/devel/lib/envsim/visionsim_node __name:=dodgeros_pilot __log:=/home/user/.ros/log/9b8562be-a8f9-11ec-8908-f9ada68fd0d8/kingfisher-dodgeros_pilot-2.log].
log file: /home/user/.ros/log/9b8562be-a8f9-11ec-8908-f9ada68fd0d8/kingfisher-dodgeros_pilot-2*.log
`
```

Sheng-Cheng, 2022-03-23:
Hi Tsuji-Kosei! I have the same issue on my computer. Did you find a solution? Thanks!

Tsuji-Kosei, 2022-03-23:
Hi!
The cause of error is that “FLIGHTMARE_PATH” had no environment variable. Please check line 47 of envsim/src/visionsim_node.cpp. Thanks!

Tsuji-Kosei, 2022-03-23:
> Hi Tsuji-Kosei! I have the same issue on my computer. Did you find a solution? Thanks!

> Hi Tsuji-Kosei! I have the same issue on my computer. Did you find a solution? Thanks!

Hi!
The cause of error is that “FLIGHTMARE_PATH” had no environment variable. Please check line 47 of envsim/src/visionsim_node.cpp. Thanks!

Tsuji-Kosei, 2022-03-23:
> Hi Tsuji-Kosei! I have the same issue on my computer. Did you find a solution? Thanks!

Hi!
The cause of error is that “FLIGHTMARE_PATH” had no environment variable. Please check line 47 of envsim/src/visionsim_node.cpp. Thanks!

gophermmq, 2022-03-24:
Thanks a lot. I also got this problem.

Sheng-Cheng, 2022-03-24:
Hi Tsuji-Kosei, this is what I have on my line 47:
getenv("~/UZH/icra22_competition_ws/src/agile_flight/flightmare") + 

But I still get the error in roslaunch. Any suggestion? Thanks!

gophermmq, 2022-03-25:
I got path correct but it shows ' [kingfisher/dodgeros_pilot-2] process has died [pid 12751, exit code -11' after lauch.

Tsuji-Kosei, 2022-03-25:
Hi!
getenv is func to get  the environment varables , so please type `"export FLIGHTMARE=~/UZH/icra22_competition_ws/src/agile_flight/flightmare`in .bashrc.
btw, `roslaunch envsim visionenv_sim_debug.launch render:=true `will help you with debug.
thanks!