uzh-rpg/agile_flight #13: ROS Bridge deactivated / uniplot

korizona, 2022-03-22:
Hi. First, thanks to your great work and efforts to hold this competition.
I am suffering some issues that the `ROS_Bridge` continuously gets deactivated even if I input the command via ROS faster than 100Hz.
The UAV in the simulator seems to fly as I expected with the command I input, but anyway it gets deactivated.

I attach the error messages as below:

```
[ INFO] [1647953337.210806394, 1647953328.236202240]: OFF command received!
[ROS Bridge]   Deactivated!

[ INFO] [1647953423.736785555, 1647953389.336202383]: OFF command received!
[ROS Bridge]   Deactivated!
publishing and latching message for 3.0 seconds
[ INFO] [1647953427.241718100, 1647953391.756202221]: Resetting simulator!
publishing and latching message for 3.0 seconds
[ INFO] [1647953430.747328953, 1647953394.196202278]: Computing active: true!
[ROS Bridge]   Deactivated!
[ROS Bridge]   Activated!
rollout_9
/home/mason/ws/competition_ws/src/agile_flight
Traceback (most recent call last):
  File "evaluation_node.py", line 12, in <module>
    from uniplot import plot
ImportError: No module named uniplot
```

Thank you in advance.
K.

P.S. I tried to install uniplot in multiple ways and I checked it is installed, but getting the same error.

lbfd, 2022-03-23:
If you have a conda environment or a virtual environment, you should be able install uniplot   (https://github.com/olavolav/uniplot) using pip as follows
```
conda activate YOUR_ENVIRONMENT
pip install uniplot
```

To check the installation, activate your environment and open a python interpreter, e.g. type `python` in the command line. Inside the python interpreter you should be able to type
```
import uniplot
```
And the command should complete without errors. If this does not work, please paste the output of 
```
pip freeze
```

engcang, 2022-03-23:
@lbfd Hi.
Do you think the `deactivated` issue from `ROS Bridge` is because of `uniplot` ???
I am suffering same issue.

engcang, 2022-03-25:
Hi. I solved the problem by changing the https://github.com/uzh-rpg/agile_flight/blob/main/launch_evaluation.bash#L43 into
```
python3 evaluation_node.py &
```
Since I was testing the code on Ubuntu 18.04 with ROS melodic.

I hope this can be helpful for anyone who is using the same versions with me.

lbfd, 2022-03-25:
Great! Since your version is also compatible with Ubuntu 20.04, I've just pushed a fix to this in commit `1c7e7e28d0762f1ac7394cafeb0032b27898a625`

closed 2022-03-25 by lbfd