uzh-rpg/agile_flight #14: Some environments are maybe buggy

ouafi98, 2022-03-22:
Hi, First thanks for your great work and your endeavor to push the boundaries of flying drones.

level: "hard" 
env_folder: "environment_0"
scene_id: 3

Maybe I'm wrong, but It seems that there may be something wrong with some environments provided.
If your look carefully at the attached video, the evaluation code running in the bottom right corner print "crashed" even
when there are no obstacles around the drone. Note that I do not modify anything in the evaluation code and since I began this challenge it's the first time (also the first environment) I encounter that behavior.

Thanks in advance to check that out.

Best regards,

https://user-images.githubusercontent.com/43729232/159560060-b866db5f-07e0-4d55-af68-dd0faa77a21a.mp4

yun-long, 2022-03-22:
thanks a lot for raising the issue. 
we will look into it. 

Best,
Yunlong

prafael18, 2022-03-24:
On a somewhat related note, I encountered a similar bug where I the drone was clearly crashing but the evaluation script wouldn't signal that there was a crash. Changing the way that the distance between the drone and the closest obstacle is computed from
```
margin = dist - obs.scale / 2
```
to
```
margin = dist - obs.scale
```
seems to fix the issue. I guess at some point `obs.scale` got changed to mean radius instead of diameter.

lbfd, 2022-03-24:
@prafael18 could you please post a screenshot/video of the situation you encountered? Note that we assume the drone to be a point, so it might look visually like it crashed but the script will only detect crashes where the drone's center collided with the sphere.

prafael18, 2022-03-25:
@lbfd I'm aware you're assuming that the drone is a point, but there are still very obvious crashes that aren't being accounted for. I recorded the video below with your original `evaluation_node.py` file. Note how there's only a single reported crash for the entire trajectory (check the end of the video for the uniplots and `summary.yaml` file).

https://user-images.githubusercontent.com/22357938/160051409-08adaf01-157d-4c16-b5d7-9c390b12a749.mp4

With my modification to change margin to `dist - obs.scale` I get a much more sensible crash report.

https://user-images.githubusercontent.com/22357938/160051726-cf74ff5d-5cc7-48fd-a24e-54e9c116b0ea.mp4

RobertoIV, 2022-03-25:
> On a somewhat related note, I encountered a similar bug where the drone was clearly crashing but the evaluation script wouldn't signal that there was a crash. Changing the way that the distance between the drone and the closest obstacle is computed from
> 
> ```
> margin = dist - obs.scale / 2
> ```
> 
> to
> 
> ```
> margin = dist - obs.scale
> ```
> 
> seems to fix the issue. I guess at some point `obs.scale` got changed to mean radius instead of diameter.

I have encountered the exact same issue,t he drone crashes and it even goes through the spheres and is not counted as a crash.  
Deleting division by 2 solved the problem.

lbfd, 2022-03-25:
@prafael18 thanks a lot for the video. Based on your observation, dist - obs.scale makes more sense. I will confirm with @yun-long but we will likely push a fix to this later today.

lbfd, 2022-03-25:
Thanks @RobertoIV @ouafi98 @prafael18 for bringing up this issue. We have fixed the issue in commit `0f775aa862347ed1ab11245a7f05d9e261111c9d`.

A bit of background: the environment files store the obstacle size as the diameter. However, the rendering engine converts them into radius and this radius is then published via ROS. The evaluation script subscribes to the ROS message and therefore did not receive the diameter (as originally expected) but the radius of the obstacle.

ouafi98, 2022-03-25:
@ibfd The error is not gone. in the same environment, I'm still getting the same behavior.
Note: I re-installed all from scratch.

ouafi98, 2022-03-25:
@ibfd 

I think I might have caught the issue. I guess it comes from the render_offset parameter in the file "scene. YAML".
I did some tests, when I set this parameter to 0 for all the problematic environments I've encountered, the issue disappear (I mean I can see myself the obstacle causing the collision)

My guesses:
-  render_offset maybe doesn't correctly do its job (I don't know what it is), maybe translating all the environment.
- in vision-based control, this is problematic since you get penalized for not avoiding something you didn't/can't see.

Hoping this will help tackle this issue.
Best

yun-long, 2022-03-27:
@ouafi98 

thanks for the feedback. the issue has been addressed. check the Update in the readme.

ouafi98, 2022-03-27:
Thanks @yun-long 

I've downloaded the new file and did a git pull, this time I can't connect to unity. Unity starts well but the code stays on trying to connect Unity. I'll try to start over or reboot my computer. I'll contact you if the error holds after that.

Best

ouafi98, 2022-03-27:
I started over. everything seems to work fine (the issue has disappeared) ^_^.

closed 2022-03-27 by ouafi98

prafael18, 2022-03-28:
Sorry to reopen the issue, but I'm now having the same problem as @ouafi98 reported.

I've already went through the entire setup again, cloning and setting up everything from scratch, but roslaunch gets stuck in the `Trying to Connect Unity` step. It's weird because everything opens, including rosviz, dodgeros_gui, and RPG_Flightmare, but the roslaunch script can't seem to connect to Flightmare. It might be worth mentioning that everything was working properly before the recent update to fix static obstacles rendering. Not sure if this helps, but here are the logs printed to stdout:

```
> [agileflight] rafaelfp@icra22_competition_ws/src/agile_flight (main|✚4…) roslaunch envsim visionenv_sim.launch render:=True

... logging to /home/rafaelfp/.ros/log/7b99c806-aea7-11ec-9394-55e58ebb87bd/roslaunch-rafaelfp-Desktop-5377.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rafaelfp-Desktop:36841/

SUMMARY
========

PARAMETERS
 * /kingfisher/dodgeros_pilot/agi_param_dir: /home/rafaelfp/Do...
 * /kingfisher/dodgeros_pilot/camera_config: /home/rafaelfp/Do...
 * /kingfisher/dodgeros_pilot/low_level_controller: Simple
 * /kingfisher/dodgeros_pilot/pilot_config: /home/rafaelfp/Do...
 * /kingfisher/dodgeros_pilot/real_time_factor: 1.0
 * /kingfisher/dodgeros_pilot/render: True
 * /kingfisher/dodgeros_pilot/ros_param_dir: /home/rafaelfp/Do...
 * /kingfisher/dodgeros_pilot/use_bem_propeller_model: False
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /use_sim_time: True

NODES
  /
    dodgeros_gui (rqt_gui/rqt_gui)
    flight_render (flightrender/RPG_Flightmare.x86_64)
  /kingfisher/
    dodgeros_pilot (envsim/visionsim_node)
    viz_face (rviz/rviz)

auto-starting new master
process[master]: started with pid [5403]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7b99c806-aea7-11ec-9394-55e58ebb87bd
process[rosout-1]: started with pid [5413]
started core service [/rosout]
process[kingfisher/dodgeros_pilot-2]: started with pid [5420]
process[kingfisher/viz_face-3]: started with pid [5421]
process[dodgeros_gui-4]: started with pid [5422]
process[flight_render-5]: started with pid [5423]
[ WARN] [1648479490.347006106]: Pilot Config:        /home/rafaelfp/Documents/unicamp/masters/offline_rl/icra22_competition_ws/src/agile_flight/envsim/parameters/simple_sim_pilot.yaml
[ WARN] [1648479490.347389278]: Agi Param Directory: /home/rafaelfp/Documents/unicamp/masters/offline_rl/icra22_competition_ws/src/agile_flight/dodgedrone_simulation/dodgelib/params
[ WARN] [1648479490.347408265]: ROS Param Directory: /home/rafaelfp/Documents/unicamp/masters/offline_rl/icra22_competition_ws/src/agile_flight/envsim/parameters
[ INFO] [1648479490.347444886]: Loading Pilot Params from /home/rafaelfp/Documents/unicamp/masters/offline_rl/icra22_competition_ws/src/agile_flight/envsim/parameters/simple_sim_pilot.yaml in /home/rafaelfp/Documents/unicamp/masters/offline_rl/icra22_competition_ws/src/agile_flight/envsim/parameters
Loading Pilot parameters from "/home/rafaelfp/Documents/unicamp/masters/offline_rl/icra22_competition_ws/src/agile_flight/envsim/parameters/simple_sim_pilot.yaml"
[PilotParams]  Did not create inner controller ''!
[Pilot]        Did not create bridge 'ROS'.
Using debug bridge, register externally!
[ INFO] [1648479490.349720196]: Loaded pipeline:
Estimator:
Type: Feedthrough
File: "/home/rafaelfp/Documents/unicamp/masters/offline_rl/icra22_competition_ws/src/agile_flight/dodgedrone_simulation/dodgelib/params/feedthrough.yaml"
Sampler:
Type: Time
File: ""
Outer Controller:
Type: GEO
File: "/home/rafaelfp/Documents/unicamp/masters/offline_rl/icra22_competition_ws/src/agile_flight/dodgedrone_simulation/dodgelib/params/geo.yaml"
Inner Controller:
Type: 
File: ""
Bridge:
Type: ROS
File: ""

[Pilot]        Register external bridge: [ROS Bridge]   
 which was not active and used.
[VisionEnv]    Camera has been added. Skipping the camera configuration.
[UnityBridge]  Initializing ZMQ connection!
[UnityBridge]  Initializing ZMQ connections done!
[VisionEnv]    Flightmare Bridge created.
[UnityBridge]  Trying to Connect Unity.
[.................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................]
[UnityBridge]  Unity Connection time out! Make sure that Unity Standalone or Unity Editor is running the Flightmare.
```
Here's a recording of what I see when trying to run roslaunch:

https://user-images.githubusercontent.com/22357938/160433285-15046d1f-1a6d-42f6-a0f7-2224d9c9149e.mp4

I can attach other log files if more info is necessary.

yun-long, 2022-03-28:
hmmmmmm.

did you 

1) git pull and catkin build
2) download the newest Unity standalone and extract it under flightmare/flightrender
3) make you there is no other unity dead process running in the background.

prafael18, 2022-03-28:
Oops! I forgot to catkin build at the end to update the ros packages 😅. Thanks for the reminder! Everything is working fine.