uzh-rpg/agile_flight #15: visionenv_sim.launch prints out yellow and red messages

gophermmq, 2022-03-24:
Thanks for the great work. When I follow the steps in the instructions:
                roslaunch envsim visionenv_sim.launch render:=true
finally shows 'UnityBridge]  Flightmare Unity is connected.'  but prints out some warning (in yellow) and red messages. I just would like to check whether it is normal. It says: 

******************  message in yellow ****************
[PilotParams]  Did not create inner controller ''!
[Pilot]        Did not create bridge 'ROS'.
Using debug bridge, register externally!
**********************************

*******message in red *****************
[dodgeros_gui-4] process has died [pid 47655, exit code -11, cmd /opt/ros/noetic/lib/rqt_gui/rqt_gui -s dodgeros_gui.basic_flight.BasicFlight --args --quad_name kingfisher __name:=dodgeros_gui __log:=/home/minghan/.ros/log/c1d886b4-ab18-11ec-b3ca-07c585af6c36/dodgeros_gui-4.log].
log file: /home/minghan/.ros/log/c1d886b4-ab18-11ec-b3ca-07c585af6c36/dodgeros_gui-4*.log
***************************************

Could I ask whether these are normal or there is a setting-up issue? Thank you very much.

gophermmq, 2022-03-24:
By the way, when I followed all the steps, the uav did not move. I guess there is something wrong in my setup.

yun-long, 2022-03-24:
Hi, 

there is something wrong with the setup. yellow messages sometimes are fine. but, the red message means the ROS node is failed.

Sheng-Cheng, 2022-03-25:
I also get this error. I think the gui is dead and then you can't command the vehicle.

[dodgeros_gui-4] process has died [pid 6730, exit code 1, cmd /opt/ros/noetic/lib/rqt_gui/rqt_gui -s dodgeros_gui.basic_flight.BasicFlight --args --quad_name kingfisher __name:=dodgeros_gui __log:=/home/sheng/.ros/log/0ee9fdd4-abeb-11ec-b44e-712162f9a126/dodgeros_gui-4.log].
log file: /home/sheng/.ros/log/0ee9fdd4-abeb-11ec-b44e-712162f9a126/dodgeros_gui-4*.log

lbfd, 2022-03-25:
Can you please post the full terminal output of the error? Before the GUI process dies?

gophermmq, 2022-03-25:
> I also get this error. I think the gui is dead and then you can't command the vehicle.
>
> [dodgeros_gui-4] process has died [pid 6730, exit code 1, cmd /opt/ros/noetic/lib/rqt_gui/rqt_gui -s [...]

gophermmq, 2022-03-25:
Here is my terminal output with errors:

*****************************************

auto-starting new master
process[master]: started with pid [55150]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9de44bb0-ac41-11ec-a0f4-833bfc0b1034
process[rosout-1]: started with pid [55160]
started core service [/rosout]
process[kingfisher/dodgeros_pilot-2]: started with pid [55167]
process[kingfisher/viz_face-3]: started with pid [55168]
process[dodgeros_gui-4]: started with pid [55169]
process[flight_render-5]: started with pid [55170]
[ WARN] [1648215836.833028841]: Pilot Config:        simple_sim_pilot.yaml
[ WARN] [1648215836.833497074]: Agi Param Directory: /home/minghan/icra22_competition_ws/src/agile_flight/dodgedrone_simulation/dodgelib/params
[ WARN] [1648215836.833518243]: ROS Param Directory: /home/minghan/icra22_competition_ws/src/agile_flight/envsim/parameters
[ INFO] [1648215836.833539656]: Loading Pilot Params from simple_sim_pilot.yaml in /home/minghan/icra22_competition_ws/src/agile_flight/envsim/parameters
Loading Pilot parameters from "/home/minghan/icra22_competition_ws/src/agile_flight/envsim/parameters/simple_sim_pilot.yaml"
[PilotParams]  Did not create inner controller ''!
[Pilot]        Did not create bridge 'ROS'.
Using debug bridge, register externally!
[ INFO] [1648215836.835993646]: Loaded pipeline:
Estimator:
Type: Feedthrough
File: "/home/minghan/icra22_competition_ws/src/agile_flight/dodgedrone_simulation/dodgelib/params/feedthrough.yaml"
Sampler:
Type: Time
File: ""
Outer Controller:
Type: GEO
File: "/home/minghan/icra22_competition_ws/src/agile_flight/dodgedrone_simulation/dodgelib/params/geo.yaml"
Inner Controller:
Type: 
File: ""
Bridge:
Type: ROS
File: ""

[Pilot]        Register external bridge: [ROS Bridge]   
 which was not active and used.
[kingfisher/dodgeros_pilot-2] process has died [pid 55167, exit code -11, cmd /home/minghan/icra22_competition_ws/devel/lib/envsim/visionsim_node __name:=dodgeros_pilot __log:=/home/minghan/.ros/log/9de44bb0-ac41-11ec-a0f4-833bfc0b1034/kingfisher-dodgeros_pilot-2.log].
log file: /home/minghan/.ros/log/9de44bb0-ac41-11ec-a0f4-833bfc0b1034/kingfisher-dodgeros_pilot-2*.log
PluginManager._discover() force discovery of plugins
RospkgPluginProvider._find_plugins() crawling for plugins of type 'qt_gui'
[dodgeros_gui-4] process has died [pid 55169, exit code -11, cmd /opt/ros/noetic/lib/rqt_gui/rqt_gui -s dodgeros_gui.basic_flight.BasicFlight --args --quad_name kingfisher __name:=dodgeros_gui __log:=/home/minghan/.ros/log/9de44bb0-ac41-11ec-a0f4-833bfc0b1034/dodgeros_gui-4.log].
log file: /home/minghan/.ros/log/9de44bb0-ac41-11ec-a0f4-833bfc0b1034/dodgeros_gui-4*.log

lbfd, 2022-03-25:
The log so far looks ok and we can't reproduce the problem. Could you please list your system details:
- ubuntu version
- kernel version
- processor you use (like AMD, Intel, ARM)
- GCC version
- python version
- Eigen version
(and anything else you think might be helpful for us).

gophermmq, 2022-03-25:
I have:
Ubuntu 20.04.4 LTS, 
kernal 5.13.0-37-generic
processor: Intel® Core™ i7-10700F CPU
NVIDIA Corporation TU116 [GeForce GTX 1660 Ti]
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
python 3.8.10
'pkg-config --modversion eigen3' prints out 3.3.7

Thanks for the help. I am also trying to figure out the issue.

k-okada, 2022-04-05:
@HarukiKozukapenguin 
It is working on my environment. First please try with only ROS, without Anaconda, so please remove anaconda settings from `~/.bashrc`
Then, follow the 1st step of instructions (https://github.com/uzh-rpg/agile_flight/blob/main/README.md)
```
roslaunch envsim visionenv_sim.launch render:=True
# Using the GUI, press Arm & Start to take off.
```
and send velocity
```
rostopic pub -r 10 /kingfisher/dodgeros_pilot/velocity_command geometry_msgs/TwistStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
twist:
  linear:
    x: 0.0
    y: 0.0
    z: 1.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0" 
```
You can see the robot is going up

Then you start different terminal, `source cra22_competition_ws/deve/setup.bash` , `conda activate agileflight0`, then start the python code.

Cc: @Takuzumi240, @tongtybj

tongtybj, 2022-04-05:
@k-okada @HarukiKozukapenguin 

I aslo succeeded to launch `visionenv_sim.launch` in Docker env like this:
[image]

I succeeded to use following topics to control the drone:
- `/kingfisher/dodgeros_pilot/start`: takeoff
- `/kingfisher/dodgeros_pilot/land`: landing
- `/kingfisher/dodgeros_pilot/go_to_pose`: position-based navigation

HarukiKozukapenguin, 2022-04-06:
@k-okada 
Thank you for helping me.
I launched by following instruction.

`roslaunch envsim visionenv_sim.launch render:=True`
The simulator launched as expected.
But, when we send velocity topic by following command, the process of launch file is dead with the following command.
```
rostopic pub -r 10 /kingfisher/dodgeros_pilot/velocity_command geometry_msgs/TwistStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
twist:
  linear:
    x: 0.0
    y: 0.0
    z: 1.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
```
```
[kingfisher/dodgeros_pilot-2] process has died [pid 3082, exit code -11, cmd /home/myplace/icra22_competition_ws/devel/lib/envsim/visionsim_node __name:=dodgeros_pilot __log:=/home/myplace/.ros/log/61ddaeb2-b57f-11ec-a1f2-dbf059b0747b/kingfisher-dodgeros_pilot-2.log].
log file: /home/myplace/.ros/log/61ddaeb2-b57f-11ec-a1f2-dbf059b0747b/kingfisher-dodgeros_pilot-2*.log
```
So, I will check in the docker env.

HarukiKozukapenguin, 2022-04-06:
When we push "Arm bridge" , then this bug is cured.

emanuelazfernandes, 2022-04-23:
I don't know if you fixed it, but the first error I was getting with:
[dodgeros_gui-4] process has died [pid 47655, exit code -11, cmd /opt/ros/noetic/lib/rqt_gui/rqt_gui -s
was solved by conda deactivate.
The second one, I suggest removing the export ROS_VERSION variable from .bashrc, as it conflicts with the one already present in /opt/ros/noetic/setup.bash
Finally, I suggest also putting the full path in the FLIGHTMARE_PATH variable, e.g /home/yourusername, so the last 3 lines of my ~/.bashrc look like this:

source /opt/ros/noetic/setup.bash
source /home/myusername/catkin_workspaces/icra22_competition_ws/devel/setup.bash
export FLIGHTMARE_PATH=/home/myusername/catkin_workspaces/icra22_competition_ws/src/agile_flight/flightmare

HarukiKozukapenguin, 2022-04-26:
I have already fixed it, and I now run in python