uzh-rpg/agile_flight #16: RL Environment Command Modes

prafael18, 2022-03-25:
Hi! I followed through with the python setup steps and was looking into training an RL policy with the provided environment using the `run_vision_demo.py` script as a reference. Looking into the environment code, it seems like only one type of action is supported, namely the CTBR actions. In contrast, the ROS setup supports all three command modes. Are there any plans to support the other action modes for the RL environment? I'm aware that your lab has found the CTBR commands to be the [best ones](https://arxiv.org/abs/2202.10796) for policy training, but I feel like this should still be an option for other participants to experiment with.

On a somewhat related note, I also had a few other concerns with the current configs for the simulation environments. I could open other issues if this is the preferred method to address these concerns.
1. I noticed that the simulation step time is 0.02s (i.e., `sim_dt` in the `flightmare/flightpy/configs/vision/config.yaml` file) by default, which amounts to 50Hz. Is this going to be fixed for the competition or are we free to modify this as we please. I feel like if we are going to use these lower level command modes like SRT and CTBR, having a higher simulation step frequency would be important.
2. I'm not entirely sure about this concern, but looking through the LINVEL command mode code it seems like it doesn't actually modify the rotor thrusts of the drone and instead just changes the current position of the drone in the environment (I'm using the `velocity_reference.cpp` file as a reference). I was wondering if there was a plan to translate these state updates to actual rotor thrusts, since they would provide more realistic movement dynamics for the drone.

Finally, I feel like a lot of questions can be answered more quickly through a different communication platform. Do you all consider creating a discord server, for instance, to facilitate these types of quick questions that one might have about the competition? I feel like having a space for all participants to discuss their hurdles and concerns would benefical for everyone.

ouafi98, 2022-03-25:
@prafael18, a discord server or something like that is a great idea!

yun-long, 2022-03-28:
hi @prafael18 

Indeed, adding support to other control modes is definitely useful. At the moment, we don't have a very specific plan for implementing more control modes. We welcome pull requests. Feel free to change the stimulation frequency.

Regarding the discord server, I never used it before.  Not sure if it will be a good option for all participants.