uzh-rpg/agile_flight #17: Pyside error?

ngthanhtin, 2022-03-28:
Hi, when I run `roslaunch envsim visionenv_sim.launch render:=True`, it has successfully opened rviz, unity,.. but still got one or two error.
1) pyside error: `ImportError: cannot import name 'libqt_gui_cpp_shiboken'`
2) error in Yaml file

[](url
[image]
)
Do  you know what the problem? thanks

ouafi98, 2022-03-28:
For the issue with the yaml file check the variable FLIGHTMARE_PATH.

ngthanhtin, 2022-03-28:
@ouafi98 I have source it in bashrc,for the 1) errror, when I run the launch_evaluation.bash, it can open like this, but the GUI for controling the drone is not working:
[image]

yun-long, 2022-03-28:
hi

1) did you resolve the YAML error?
2) which python version are you using? it seems there is a problem with the python GUI. when you click connect on the GUI, does it work?

ngthanhtin, 2022-03-28:
@yun-long i havent solved the yaml file error.
I use python3.6, the gui does not have any connect button, it only has off red button like the above image?

yun-long, 2022-03-28:
1) can you check what is the output when you do 
`
 echo $FLIGHTMARE_PATH 
// the output should be 
/something/src/agile_flight/flightmare
`

2) can you try to config your python executable to the python3
`
cd <your catkin work space>
catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3
`

gorghino, 2022-03-28:
> @yun-long the gui does not have any connect button, it only has off red button like the above image?

Did you try the 'Connect' button above the red square?

ngthanhtin, 2022-03-29:
@gorghino yes I tried but nothing happened

ngthanhtin, 2022-03-29:
@yun-long yes I sourced exactly the my FLIGHTMARE_PATH.
I set up my workspace to work with python3 as well

ngthanhtin, 2022-03-29:
When I click on the GUI the cmd output like this:
[image]

ouafi98, 2022-03-29:
maybe try to upgrade your python to at least 3.7
I have 3.8 it works just fine for me.