uzh-rpg/agile_flight #18: Change environment/scene during training

ouafi98, 2022-03-29:
Hi
I've spent some time finding a way to change the environment/scene during the training I didn't find anything easy.
I've found addDynamicObstacle (but not its counterpart ), also found addStaticObsacle but it seems that it is not used anywhere.

Do you have some tips on How I can achieve that? (I want to randomly appear in an environment/scene during training) 

Best r

yun-long, 2022-03-30:
Hi 

It is very difficult to change the environment during the training since Unity itself does not support it.

ouafi98, 2022-03-30:
Thanks for the reply.

What about continuing learning / curriculum learning in rpg_baselines ?

Can I save a model with model.save()
Then load it for training in another environment and be sure to not start from scratch ?

Br.

yun-long, 2022-03-31:
yes. this is for sure possible. 

For this, I suggest reading the documents about stable baselines3

https://stable-baselines3.readthedocs.io/en/master/index.html

closed 2022-04-01 by ouafi98