uzh-rpg/agile_flight #21: obstacle avoidance?

ngthanhtin, 2022-04-01:
Hi i would like to ask how to adjust the env for  terminating when encounter an obstacle.
In vision_env.cpp, I have changed the code as following:
[image]
and then,
`catkin build flightlib`
But the agent still is going through obstacles as the following video:
https://www.youtube.com/watch?v=350ardLDI1Q&ab_channel=T%C3%ADnNguy%E1%BB%85n

So could you tell me how to do this?

Thanks

yun-long, 2022-04-01:
this code is used for reinforcement learning, which does not use any ROS. 

when you change the vision_env.cpp, you should do

```
cd flightlib/build
cmake ..
make -j10
pip install . 
```

by doing so, the flightgym python package will be updated. 
the flightgym package is used in the (run_vision_ppo.py)[https://github.com/uzh-rpg/agile_flight/blob/main/envtest/python/run_vision_ppo.py]

ngthanhtin, 2022-04-01:
Thanks, I did it.

closed 2022-04-01 by ngthanhtin