uzh-rpg/agile_flight #26: Could not import "pyqt" bindings of qt_gui_cpp library

arrijon96, 2022-04-06:
Hi,
After following completing the installation, when running `roslaunch envsim visionenv_sim.launch render:=True`, I get the following output:
```
...
[VisionEnv]    Camera has been added. Skipping the camera configuration.
[UnityBridge]  Initializing ZMQ connection!
[UnityBridge]  Initializing ZMQ connections done!
[VisionEnv]    Flightmare Bridge created.
[UnityBridge]  Trying to Connect Unity.
[.......Could not import "pyqt" bindings of qt_gui_cpp library - so C++ plugins will not be available:
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ImportError: /home/name/anaconda3/lib/python3.9/site-packages/PyQt5/../../../libQt5Core.so.5: version `Qt_5.12' not found (required by /opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/libqt_gui_cpp_sip.so)
............................................................................................................................................................[UnityBridge]  Flightmare Unity is connected.
...
```
Ignoring the `Could not import "pyqt" bindings of qt_gui_cpp library...` message, everything seems to be fine. However, when pressing `Connect` and `Arm` in the GUI, I get the following errors:
```
[ INFO] [1649252353.976127531, 1649252345.667026281]: Computing active: true!
[ROS Bridge]   Deactivated!
[ROS Bridge]   Activated!
[Pipeline]     Bridge failed!
[ROS Bridge]   Deactivated!
[Pipeline]     Bridge failed!
[Pipeline]     Bridge failed!
[Pipeline]     Bridge failed!
[Pipeline]     Bridge failed!
[Pipeline]     Bridge failed!
[Pipeline]     Bridge failed!
...
```
Unfortunately, this `[Pipeline]     Bridge failed!` message keeps running unless I kill the node(s).

I am looking forward to your answer,

Jon

arrijon96, 2022-04-06:
Hi,
I just realized that this behavior was caused by the fact that I was running it in a virtual machine (virtualbox). As soon as I replicated the installation within native Ubuntu, everything worked out of the box. Sorry for the mess. I will leave this issue opened, since it might still be relevant for running the framework in VMs.
Thanks,
Jon

yun-long, 2022-04-06:
if you are using anaconda, you can try to specify the python executable in the catkin workspace

```
cd your_catkin_ws
catkin config -DPYTHON_EXECUTABLE=/home/yunlong/anaconda3/envs/flight/bin/python3
catkin clean
caktin build
```