uzh-rpg/agile_flight #33: unable to run visionenv_sim.launch file

deepak-1530, 2022-04-08:
After following the instruction given in the README.md, I ran the command roslaunch envsim visionenv_sim.launch render:=True, the flightmare simulator and rviz windows opened up but also got the following error -:

[dodgeros_gui-4] process has died [pid 2562, exit code 1, cmd /opt/ros/melodic/lib/rqt_gui/rqt_gui -s dodgeros_gui.basic_flight.BasicFlight --args --quad_name kingfisher __name:=dodgeros_gui __log:=/home/deepak/.ros/log/cc19a17a-b780-11ec-b7cf-a4b1c1152af2/dodgeros_gui-4.log].
log file: /home/deepak/.ros/log/cc19a17a-b780-11ec-b7cf-a4b1c1152af2/dodgeros_gui-4*.log

[ INFO] [1649452436.562074185]: Loading Pilot Params from simple_sim_pilot.yaml in /home/deepak/DodgeDrone/src/agile_flight/envsim/parameters
Loading Pilot parameters from "/home/deepak/DodgeDrone/src/agile_flight/envsim/parameters/simple_sim_pilot.yaml"
terminate called after throwing an instance of 'agi::ParameterException'
  what():  Dodgelib Parameter Exception: Quadrotor file is set manually an in YAML!

lbfd, 2022-04-11:
We cannot reproduce this on our end, are you sure that you don't have local changes? It seems you changed the path to the quadrotor yaml file. 

In general, we can't really support all the different ROS versions, so please make sure that you are on Ubutu 20.04 with ROS noetic and a recent Python+GCC version.

xingcw, 2022-04-12:
> After following the instruction given in the README.md, I ran the command roslaunch envsim visionenv_sim.launch render:=True, the flightmare simulator and rviz windows opened up but also got the [...]

I got the same problem on ubuntu 20.04. I found it was because the "\r"s in the simple_sim_pilot.yaml file. A quick solution is to convert that file to "UNIX" format using, for example, notepad.