uzh-rpg/agile_flight #36: ModuleNotFoundError: No module named 'flightrl'

HarukiKozukapenguin, 2022-04-09:
Thank you for interesting challenge.
I ran python.run_vision_ppo with the folloing command, but the command says there are no module named flightrl'
$ python3 -m python.run_vision_ppo --render 0 --train 1
Traceback (most recent call last):
  File "/home/myworkspace/anaconda3/envs/agileflight/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/hamyworkspaceu/anaconda3/envs/agileflight/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/myworkspace/icra22_competition_ws/src/agile_flight/envtest/python/run_vision_ppo.py", line 14, in <module>
    from rpg_baselines.torch.common.ppo import PPO
  File "/home/myworkspace/anaconda3/envs/agileflight/lib/python3.8/site-packages/rpg_baselines/torch/common/ppo.py", line 21, in <module>
    from flightrl.rpg_baselines.torch.common.on_policy_algorithm import \
ModuleNotFoundError: No module named 'flightrl'

I did the following things.(cf. https://github.com/uzh-rpg/agile_flight/blob/main/envtest/python/README.md)

- Run _./setup_py.bash_ with conda environment.
- Checked to run well the $ python3 -m python.run_vision_demo --render 1

I have several bugs when run _./setup_py.bash_ , and I cured these by myself.
If these cause some problem, prease tell me
A. Based on this issue, I changed f
lightmare/flightpy/flightrl/setup.py from
packages=['rpg_baselines'],
to
packages=find_packages()
https://github.com/uzh-rpg/agile_flight/issues/5#issuecomment-1072016802

tongtybj, 2022-04-09:
@HarukiKozukapenguin 

I got the same error with you.
The reason is that the package of `rpg_baselines` is installed without the upper prefix of `flightrl` (Please see https://github.com/uzh-rpg/flightmare/blob/master/flightrl/setup.py).

However, https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/rpg_baselines/torch/common/ppo.py#L21-L23 uses a prefix of `flightrl`. 

A simple workaround is to change https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/rpg_baselines/torch/common/ppo.py#L21-L23
as follows:

```
from rpg_baselines.torch.common.on_policy_algorithm import \
    OnPolicyAlgorithm
from rpg_baselines.torch.common.util import plot3d_traj, traj_rollout
```
and redo https://github.com/uzh-rpg/agile_flight/blob/main/setup_py.bash#L64-L66 in your conda enviroment.

tongtybj, 2022-04-09:
@yun-long 

What is the purpose to add `flightrl` in https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/rpg_baselines/torch/common/ppo.py#L21-L23?

This breaks the consistence with your rule described in https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/setup.py#L19

yun-long, 2022-04-11:
hi @tongtybj 

thanks a lot for helping to address the issues. 

You are right. there should be no flightrl in the ppo.py

HarukiKozukapenguin, 2022-04-12:
@tongtybj 
Thank you so much.
I fixed, and it works well.

closed 2022-04-14 by HarukiKozukapenguin