uzh-rpg/agile_flight #45: goal_vel ?

HarukiKozukapenguin, 2022-04-14:
[The goal is to proceed as fast as possible 60m in positive x-direction without colliding into obstacles and exiting a pre-defined bounding box](https://github.com/uzh-rpg/agile_flight/blob/main/README.md#testing-todo). Then, what the [`goal_vel`](  goal_vel: [3.0, 0.0, 0.0] # goal velicty for tracking vx, vy, vz) stands for? Should drone be such velocity when we reach the goal?, Or there are no restriction for velosity?

yun-long, 2022-04-15:
it is a hyperparameter that you can change

you can use adaptive goal velocity depending on how the distance between the current quadrotor and the goal position/

or, you don't need a goal velocity but minimize the distance between the current quadrotor and the goal positon.

HarukiKozukapenguin, 2022-04-15:
Does _adaptive goal velocity_ mean "desired velocity for quadrotor"?
I think so because quadrotor [get penalty depending on gap between velocity of quadrotor and `goal_vel`](https://github.com/uzh-rpg/flightmare/blob/ee30f203df42596668ee4c386ce1e45aedeedb8c/flightlib/src/envs/vision_env/vision_env.cpp#L313-L314).

closed 2022-04-27 by HarukiKozukapenguin