uzh-rpg/agile_flight #46: Changing environment

ddirckx, 2022-04-14:
Following the description in the Readme, we tried changing the environment level in [this file](https://github.com/uzh-rpg/flightmare/blob/dev/version_22/flightpy/configs/vision/config.yaml) and setting (unity scene_id) to easy/medium/hard and to all possible environments. However, when we launch the simulator using the given roslaunch command (roslaunch envsim visionenv_sim.launch render:=True), the environment has not changed.

Are we missing some parameters that we additionally need to set to change the environment in Rviz and in Flightmare ?

Thanks in advance!

yun-long, 2022-04-15:
so far I can tell you are changing the right file. 

can you double-check if you followed all the installation steps.

ddirckx, 2022-04-19:
Thanks for the quick reply! We found the error ourselves. We still had a different installation meaning that the FLIGHTMARE_PATH was set to the old installation and hence could not find the correct yaml configuration.