uzh-rpg/agile_flight #47: Data interpretation logged by tensorboard_log

HarukiKozukapenguin, 2022-04-15:
Thank you for interesting simulator!

I checked `run_vision_ppo.py` by following command.
`python3 -m python.run_vision_ppo --render 0 --train 1` 
And, I found data when they train in the `envtest/python/saved` directory (e.g. PPO_1, PPO_2). I found some policies when they're training(/policy), and Test Trajectory(/TestTraj).
The questions I would like to ask is as follows.
1. where is the logging of reward transition when they train?
2. what does each axis means in the graph of  TestTraj/Plots?
3. where is the parameter of plotting or logging?

yun-long, 2022-04-15:
hi, 

1) the training reward is logged in tensorboard. you can to to saved and run

`
tensorboard --logdir=./
`

2) the plots in the first row are position [x, y, z] and the plots in the second row are velocity [x, y, z]

3) the plotting is done [here](https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/rpg_baselines/torch/common/ppo.py#L335-L375)

HarukiKozukapenguin, 2022-04-15:
1. Which directory should I run this command `tensorboard --logdir=./, or should I run this command b/f I run `python3 -m python.run_vision_ppo --render 0 --train 1`?

HarukiKozukapenguin, 2022-04-16:
I move to a directory of saved/PPO_(num) and I run `tensorboard --logdir=./` then I can seer transition of each reward.
Thank you!

HarukiKozukapenguin, 2022-04-18:
@yun-long 

I have one question about interpretation of TensorBoard.
I can see rewards transition when I learns in the simulation, but I do not know how what does it means.

1. I think the reward is a sum of reward of each episode, Is it correct?
2. Is this rewards a rewards when agent **trains**? or rewards when agent **evaluate**?

yun-long, 2022-04-18:
Hi, 

the reward you see on Tensorboard is from [here](https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightlib/src/envs/vision_env/vision_env.cpp#L325-L326). 

In summary, contains the sum reward and each individual reward component. The reward is a training reward, not an evaluation reward.

HarukiKozukapenguin, 2022-04-18:
Which code does write these rewards to TensorBoard?

yun-long, 2022-04-20:
hi 

the code is [here](https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/rpg_baselines/torch/common/on_policy_algorithm.py#L294-L326)

closed 2022-04-21 by HarukiKozukapenguin