uzh-rpg/agile_flight #48: Goal check in RL

HarukiKozukapenguin, 2022-04-16:
Thank you for interesting challenge!

I tried to learn quadrotor to move well.
I read code of [`vision_env.cpp`](https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightlib/src/envs/vision_env/vision_env.cpp) and [`config.yaml`](https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/configs/vision/config.yaml) but there are no information about goal position (+60) where is this information or should I input this information when I want to use this information?

yun-long, 2022-04-16:
the vision_env.cpp is just an example. you should change the code. 
for example, you have to define the goal position or goal velocity.
you should define terminal conditions. For example, when reaching the goal position, the episode is terminated.

HarukiKozukapenguin, 2022-04-16:
OK. I will manage by myself.

closed 2022-04-16 by HarukiKozukapenguin