uzh-rpg/agile_flight #49: Unable to build flightlib

yukaryote, 2022-04-19:
When I try to build flightlib with
```
cd flightlib/build
cmake ..
make -j10
```
I get the following error on the `make -j10` step: 
```
/usr/bin/ld: libflightlib.a(vec_env_base.cpp.o): in function `flightlib::VecEnvBase<flightlib::QuadrotorEnv>::VecEnvBase()':
vec_env_base.cpp:(.text._ZN9flightlib10VecEnvBaseINS_12QuadrotorEnvEEC2Ev[_ZN9flightlib10VecEnvBaseINS_12QuadrotorEnvEEC5Ev]+0x11b): undefined reference to `omp_set_num_threads'
/usr/bin/ld: libflightlib.a(vec_env_base.cpp.o): in function `flightlib::VecEnvBase<flightlib::QuadrotorEnv>::configEnv(YAML::Node const&)':
vec_env_base.cpp:(.text._ZN9flightlib10VecEnvBaseINS_12QuadrotorEnvEE9configEnvERKN4YAML4NodeE[_ZN9flightlib10VecEnvBaseINS_12QuadrotorEnvEE9configEnvERKN4YAML4NodeE]+0x6ec): undefined reference to `omp_set_num_threads'
/usr/bin/ld: libflightlib.a(vec_env_base.cpp.o): in function `flightlib::VecEnvBase<flightlib::VisionEnv>::VecEnvBase()':
vec_env_base.cpp:(.text._ZN9flightlib10VecEnvBaseINS_9VisionEnvEEC2Ev[_ZN9flightlib10VecEnvBaseINS_9VisionEnvEEC5Ev]+0x11b): undefined reference to `omp_set_num_threads'
/usr/bin/ld: libflightlib.a(vec_env_base.cpp.o): in function `flightlib::VecEnvBase<flightlib::VisionEnv>::configEnv(YAML::Node const&)':
vec_env_base.cpp:(.text._ZN9flightlib10VecEnvBaseINS_9VisionEnvEE9configEnvERKN4YAML4NodeE[_ZN9flightlib10VecEnvBaseINS_9VisionEnvEE9configEnvERKN4YAML4NodeE]+0x6ec): undefined reference to `omp_set_num_threads'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_lib.dir/build.make:287: test_lib] Error 1
make[1]: *** [CMakeFiles/Makefile2:193: CMakeFiles/test_lib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 85%] Linking CXX shared module flightgym.cpython-38-x86_64-linux-gnu.so
[ 85%] Built target flightgym
make: *** [Makefile:141: all] Error 2
```

I did add functions to vision_env.cpp and vision_vec_env.cpp, but I didn't change any of the omp related code

yun-long, 2022-04-20:
can you try to remove everything in build folder, except the setup.py file 
and try again
```

cmake ..
make -j10 

```

yukaryote, 2022-04-21:
Ah that fixed it, thanks!

closed 2022-04-21 by yukaryote