uzh-rpg/agile_flight #50: Which code desines "itaration" in eval funcion in ppo.py when learning

HarukiKozukapenguin, 2022-04-19:
Thank you for interesting challenge!

I know recording of Policy, RMS, and Test Traj is conducted in [`eval` function](https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/rpg_baselines/torch/common/ppo.py#L335-L379) .

I would like you to ask following questions.

1. Which code defines how often `eval` function is called?
2. When I learned with PPO, the iteration with `total_timesteps=int(5 * 1e7)` the iteration always finish at `iter_02000`, please tell me why this iteration number does not change
3. Does "iteration" mean how many **episodes** the agent learned? Or how many **learning iteration** agent learned?

yun-long, 2022-04-20:
hi

1.  The eval function is called [here](https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/rpg_baselines/torch/common/on_policy_algorithm.py#L342)

2.  The total number of iterations can be computed by dividing the total time steps by the total number of sampled time steps collected at each iteration. check [here](https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/rpg_baselines/torch/common/on_policy_algorithm.py#L281)

3. see 2.

HarukiKozukapenguin, 2022-04-21:
Thank you answering questions.

2. I think the log interval is defined [here](https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightpy/flightrl/rpg_baselines/torch/common/on_policy_algorithm.py#L334) So, I think the log interval is every 10 steps in this code, but I found the log interval of each episode is 50 when I run the code, So, could you tell me why this happens?

closed 2022-04-27 by HarukiKozukapenguin