uzh-rpg/agile_flight #56: Config.yaml and control structure

fudavd, 2022-04-21:
Hi,
I've been trying to hoover the quadrotor using SRT control and applying thrust proportional to the gravitational force based on the parameters of `flightpy/configs/vision/config.yaml`.
I start the simulator through `launch_evaluation.bash` and change the commands in `user_code.py`
It seems like a different drone is simulated namely: `envsim/parameters/quads/kingfisher.yaml`
I checked by changing the weight and it does affect the dynamics.

Is this the intended behaviour? Maybe I did something wrong during the install and can always copy/paste the quadrotor into the other yaml? I was wondering what drone will be used during the challenge?

On a seperate note. It seems that the user_code.py only sends signals every 0.04s (25Hz). Is it possible to increase the command request speed and/or provide a chain of commands as an output of `user_code.py` to increase the control frequency. If so where would be a good place to adapt the code?

Thank you,
Fuda

yun-long, 2022-04-22:
Hi Fuda,

yes. there are two simulators, each has its own configuration. 
The simulation in Flightmare is used for RL training because we need to parallelize the simulator. 
Another simulator is the dodgedrone simulation, which is used for ROS and for the final evaluation. 
You should not change the configuration file for the drone dynamics.

fudavd, 2022-04-22:
Thnx for the reply,
Im not sure if I understand correctly?
Why do we have the different drone dynamics?
Are we supposed to train on a different drone than the one that we are assessed on? 

Thank you!
Fuda