uzh-rpg/agile_flight #58: Flightmare-Unity time out.

Song-z-h, 2022-04-24:
Hi, 
I would like to ask that every time we do connectUnity() with an environment, Flightmare is connected with Unity and that's fine.  However, during the training phase with PPO when we stop calling render() for a few seconds, Unity disconnects from Flightmare. 
One way to avoid being disconnected is spamming render() continuously though is a bad solution. 
So is there a way to tell Unity to stay connected with Flightmare? (any parameters that I can pass to RPG_Flightmare.x86_64)

Thanks in advance, 
Greetings.

Song-z-h, 2022-04-24:
Such as if I add a time.sleep(10) inside [](https://github.com/uzh-rpg/agile_flight/blob/main/envtest/python/run_vision_demo.py) after step(), Unity will soon disconnect. the log file:
`OPENGL LOG: Creating OpenGL 4.5 graphics device ; Context level  <OpenGL 4.5> ; Context handle 10365896
Begin MonoManager ReloadAssembly
- Completed reload, in  0.069 seconds
Default vsync count 1
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 1920 x 1080 @ 60 Hz
The referenced script on this Behaviour (Game Object '!ftraceLightmaps') is missing!
UnloadTime: 1.674557 ms
Configuring sockets.
Unloading 6 Unused Serialized files (Serialized files now loaded: 0)
Trying to connect to: 127.0.0.1
Terminated ZMQ sockets.
Configuring sockets.
tcp://127.0.0.1:10253
Sockets bound.
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 1920 x 1080 @ 60 Hz
UnloadTime: 0.152963 ms
Unloading 4 unused Assets to reduce memory usage. Loaded Objects now: 2437.
Total: 0.960231 ms (FindLiveObjects: 0.092302 ms CreateObjectMapping: 0.020429 ms MarkObjects: 0.839767 ms  DeleteObjects: 0.007437 ms)

Destory Time Line
Unloading 4 Unused Serialized files (Serialized files now loaded: 0)
UnloadTime: 0.588309 ms
Unloading 22 unused Assets to reduce memory usage. Loaded Objects now: 1954.
Total: 0.836167 ms (FindLiveObjects: 0.079781 ms CreateObjectMapping: 0.019425 ms MarkObjects: 0.703432 ms  DeleteObjects: 0.033221 ms)

resize screen
requesting resize 84 x 84
resizing window to 84 x 84
Desktop is 1920 x 1080 @ 60 Hz
instantiate object
Load CSV Files
vehicle id : quadrotor0
Disable extra camera!
quadrotor0_0
0.00000	0.00000	1.00000	0.00000
0.00000	1.00000	0.00000	0.30000
-1.00000	0.00000	0.00000	0.00000
0.00000	0.00000	0.00000	1.00000
set camera view ports
set camera post process settings
Unloading 3 Unused Serialized files (Serialized files now loaded: 0)
Unloading 0 unused Assets to reduce memory usage. Loaded Objects now: 2425.
Total: 7.951463 ms (FindLiveObjects: 0.133652 ms CreateObjectMapping: 0.028125 ms MarkObjects: 7.783705 ms  DeleteObjects: 0.005340 ms)

The referenced script on this Behaviour (Game Object '!ftraceLightmaps') is missing!
Unloading 3 Unused Serialized files (Serialized files now loaded: 0)
UnloadTime: 3.638100 ms
Unloading 91 unused Assets to reduce memory usage. Loaded Objects now: 1211.
Total: 1.422194 ms (FindLiveObjects: 0.065649 ms CreateObjectMapping: 0.018372 ms MarkObjects: 1.058607 ms  DeleteObjects: 0.279282 ms)

Configuring sockets.
Trying to connect to: 127.0.0.1
Terminated ZMQ sockets.
Configuring sockets.
tcp://127.0.0.1:10253
Sockets bound.
Unloading 4 Unused Serialized files (Serialized files now loaded: 0)
requesting resize 1024 x 768
resizing window to 1024 x 768
Desktop is 1920 x 1080 @ 60 Hz
UnloadTime: 0.129247 ms
Unloading 2 unused Assets to reduce memory usage. Loaded Objects now: 2468.
Total: 1.085503 ms (FindLiveObjects: 0.109685 ms CreateObjectMapping: 0.025276 ms MarkObjects: 0.929530 ms  DeleteObjects: 0.020492 ms)
`

Song-z-h, 2022-04-26:
[image]
Is there a way to change this default time-out?

gorghino, 2022-04-27:
Hi Song,
apparently the disconnection is due to [this](https://github.com/uzh-rpg/flightmare_unity/blob/203351ffca0bf10ea64193456b3505abe9bfef85/Assets/Flightmare/Scripts/CameraController.cs#L263) strange 10s timeout introduced in FlightGoogles and kept in Flightmare by the developers.
Unfortunately _connection_timeout_seconds_default_ cannot be edited since it's defined in the standalone build.

This is even affecting my group and the disconnection is very annoying during the training phase.
Please @yun-long @tongtybj @lbfd add it as customizable parameter instead of leave it hardcoded.

yun-long, 2022-05-02:
I will update the standalone soon.

gorghino, 2022-05-07:
Up @yun-long @antonilo