uzh-rpg/agile_flight #59: Toggle visibility of drones

Song-z-h, 2022-04-24:
Hi, 
During the training of PPO, if we set num_envs to 100, There will be 100 drones in the same scene, each drone is visible to the others. That would be a problem if we train in a scene with more drones.  How could we make the drone invisible to the other drones? 
(by modifying the texture or something)

Greetings.

yun-long, 2022-05-02:
hi,

you can reduce the quadrotor size

https://github.com/uzh-rpg/flightmare/blob/092ff357139b2e98fc92bcdee50f38f85b55246d/flightlib/src/envs/vision_env/vision_env.cpp#L40

e.g., 

quad_ptr_->setSize(Vector(0.01, 0.01, 0.01));

Glanfaloth, 2023-01-02:
After setting the quadrotor size if you get "error: missing template arguments" when running make -j10 try this:
```
Vector<3> quad_size = Vector<3>(0.01, 0.01, 0.01);
quad_ptr_->setSize(quad_size);
```