uzh-rpg/agile_flight #62: Domain randomization

HarukiKozukapenguin, 2022-04-27:
Thank you for interesting challenge!

As you can see here, from the Simulation to the Real World, Domain Randomization is helpful. https://arxiv.org/abs/1703.06907

So, I would like to ask how about the quadrotor dynamics in Dodge drone challenge. Do these parameters will change in the Selections and Finals? Or, do these parameters do not change?

yun-long, 2022-05-02:
the dynamic parameters will stay the same during the evaluation. 
The only difference is that during training (if you are using Reinforcement learning), you are using a perfect simulation (no delay, no noise). 

During the evaluation, we will be using a ROS script. The ROS message publication and subscriber will have some delay, which is negligible normally. But, the performance might vary during evaluation. 

But, to answer your question. the dynamic parameters will not be changed. Not sure if Domain randomization will be a good idea in this case.

HarukiKozukapenguin, 2022-05-02:
OK, I will check these delay, Thank you for your valuable information!

closed 2022-05-02 by HarukiKozukapenguin