uzh-rpg/agile_flight #70: updateUnity extremely slow for vector environments

prafael18, 2022-05-04:
Hey everyone!

I was trying to train an agent using the depth images with the default SAC implementation from stable-baselines3 and noticed that training was extremely slow. After doing some profiling, I noticed that the bottleneck was in the updateUnity call, which is needed to retrieve the depth images. I was wondering if this was expected or if I'm doing something wrong that might be making this call take longer than expected.

By the way, I'm running the RPG_Flightmare standalone on a GPU, which I made sure of by checking with nvidia-smi, and I'm training with `num_envs = 25`. In the Flightmare [paper](https://arxiv.org/pdf/2009.00563.pdf), you guys advertise rendering speeds of up to 230Hz on a laptop, which I assume is achieved for a single environment (i.e. `num_envs = 1`). Therefore, I find it strange that I'm getting speeds closer to 1.6Hz on a GPU, even if for a larger number of environments. For reference, `updateUnity` runs at approximately 16Hz for `num_envs = 1` on my GPU.

These are my PC specs:
```
CPU: AMD Ryzen 9 3900X 12-Core Processor
GPU: GeForce RTX 2070 SUPER
RAM: 32GiB DDR4 2133 MHz
```

And these are my profiling results:
```
env.wrapper.step() took 0.42ms
env.wrapper.updateUnity() took 616.98ms
env.get_info() took 1.24ms
env.get_depth_image() took 0.04ms
env.get_obs() took 0.05ms
sac.collect_rollouts() took 621.87ms
sac.train() took 39.49ms
```

antonilo, 2022-05-07:
You might want to decrease the image resolution if you train with depth images from scratch

prafael18, 2022-05-07:
I forgot to mention, but these results were for an 84x84 depth image, which is the resolution used by the original DQN network.

Lu-tju, 2022-05-21:
Hi,
I have met the same problem as you. I run on the server with Xeon(R) CPU E5-2620 and 4 * GeForce RTX 2080. I find that the CPU, GPU, and disk io take up very little, but the rendering of unity is very slow. On my own computer, the unity works well and CPU can consume up to 100%, it looks as if the server does not play a role. Have you solved this problem? 
thanks a lot!