uzh-rpg/agile_flight #71: What scene will be used in the challenge?

MichaelBosello, 2022-05-04:
As only spheres will be considered as obstacles in the challenge, do you plan to test only in scene 0 (warehouse)? Because in scene 1 (street), the drone may spawn inside or close to a building which may mislead the agents that are based on depth camera. Also in the forest, the trees will not be considered but the agent will still see them.

antonilo, 2022-05-07:
Only spheres in the warehouse environment will be considered for the challenge. There will be no other obstacles.

ykeuter, 2022-05-07:
Thanks for this Q and A. Could you also provide any more details on the level and scoring during evaluation? I.e. should we focus on finishing a medium level as fast as possible, or focus on finishing a hard level at all (without hitting obstacles)?

closed 2022-05-17 by MichaelBosello