uzh-rpg/agile_flight #80: unexpected keyword argument 'policy_base' when running python3 -m python.run_vision_ppo --render 0 --train 1

Glanfaloth, 2022-10-28:
Hi, when I run the policy training via `python3 -m python.run_vision_ppo --render 0 --train 1`, I got this;

```
[VecEnvBase]   Load Unity configuration.
[VecEnvBase]   Load Simulation configuration.
[VecEnvBase]   100 vectorized enviromnets created. 
Vectorized Environment:
obs dim    =            [55]
act dim    =            [4]
rew dim    =            [5]
einfo dim  =            [0]
img width  =            [320]
img height =            [240]
num_envs   =            [100]
num_thread =            [10]
seed       =            [1]
scene_id   =            [0]
[VecEnvBase]   Load Unity configuration.
[VecEnvBase]   Load Simulation configuration.
[VecEnvBase]   1 vectorized enviromnets created. 
Vectorized Environment:
obs dim    =            [55]
act dim    =            [4]
rew dim    =            [5]
einfo dim  =            [0]
img width  =            [320]
img height =            [240]
num_envs   =            [1]
num_thread =            [10]
seed       =            [1]
scene_id   =            [0]
Traceback (most recent call last):
  File "/local/home/yeltao/anaconda3/envs/agileflight/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/local/home/yeltao/anaconda3/envs/agileflight/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/local/home/yeltao/thesis_ws/agile_flight/envtest/python/run_vision_ppo.py", line 125, in <module>
    main()
  File "/local/home/yeltao/thesis_ws/agile_flight/envtest/python/run_vision_ppo.py", line 74, in main
    model = PPO(
  File "/local/home/yeltao/thesis_ws/agile_flight/flightmare/flightpy/flightrl/rpg_baselines/torch/common/ppo.py", line 108, in __init__
    super(PPO, self).__init__(
  File "/local/home/yeltao/thesis_ws/agile_flight/flightmare/flightpy/flightrl/rpg_baselines/torch/common/on_policy_algorithm.py", line 80, in __init__
    super(OnPolicyAlgorithm, self).__init__(
TypeError: __init__() got an unexpected keyword argument 'policy_base'
```

How do I fix this? Thank you!

Glanfaloth, 2022-11-01:
Solved! It turns out that in the release 1.6.0 of stable-baselines3 there are some breaking changes so we have to change many arguments in the __init__()

For future students reading this: change the on_policy_algorithm.py and ppo.py according to [this](https://github.com/DLR-RM/stable-baselines3/blob/master/stable_baselines3/common/on_policy_algorithm.py) and [this](https://github.com/DLR-RM/stable-baselines3/blob/master/stable_baselines3/ppo/ppo.py), then modify the run_vision_ppo.py respectively

closed 2022-11-01 by Glanfaloth

maokat12, 2023-04-24:
Hi, I'm running into the same error message. Can I ask how you fixed it?

Glanfaloth, 2023-04-24:
Hi, you can fix it in on_policy_algorithm.py according to the error messages.

yifeichen2024, 2024-07-24:
> Hi, you can fix it in on_policy_algorithm.py according to the error messages.

Hi i have the same error. I am not quite understand your solution.